RehabMan/OS-X-BrcmPatchRAM #3: IOProviderClass changed in 10.11

dhnguyen92, 2015-06-14:
In 10.11, the IOProviderClass for Broadcom Bluetooth has changed from IOUSBDevice to IOUSBHostDevice, which causes the current bluetooth injector/brcmpatchram unsable. I changed the class in bluetooth injector kext and bluetooth works again. I tried changing in BrcmPatchRam but it is not accepted:
Jun 14 09:56:33 localhost kernel[0]: BrcmPatchRAM: Provider is not a USB device.

RehabMan, 2015-06-17:
I will look at this when I have a chance to install 10.11.  It is not surprising.

RehabMan, 2015-06-21:
If you change the IOProviderClass for BrcmPatchRAM personality, the kext stops working (because it can't cast the provider to IOUSBDevice*). As such, it simply exits without loading firmware, and you'd need a separate injector kext to provide the broadcom BT personality.

Looks like BrcmPatchRAM needs to be re-written to use IOUSBHostDevice instead of IOUSBDevice in order to work properly. They have a legacy tree in IOResources, but it appears to interfere with kexts that use the new interfaces (eg. can't have legacy kext such as BrcmPatchRAM attached to legacy tree and new kext such as BroadcomBluetoothHostControllerUSBTransport attached to the same device in the non-legacy tree). It would probably work if BrcmPatchRAM exited during probe (like BTFirmwareUploader), but it can't do that as it needs to watch power states.

Moving to IOUSBHostDevice from IOUSBDevice doesn't look difficult, but it is not trivial either.

RehabMan, 2015-06-22:
Looking into this deeper, I think what I see happening... is that the native bluetooth drivers (BroadcomBluetoothHostControllerUSBTransport.kext) now have the capability to upload firmware. As a result, BrcmPatchRAM is not required.

dhnguyen92, 2015-06-22:
Can you give me strcture for a firmware upload by BroadcomBluetoothHostControllerUSBTransport? I can't see any clue in info.plist.

RehabMan, 2015-06-22:
Haven't looked into it, but at 29kb for Info.plist, it is clear the firmware is not there.  Must be in some other component (likely a daemon in user space).  The binary doesn't look large enough to have the firmware embedded either.

RehabMan, 2015-06-22:
Note: Added BrcmBluetoothInjector.kext to the project.  It is a copy of the current BrcmPatchRAM devices, but without the firmware stuff and no code.  Just an injector (codeless kext).

RehabMan, 2015-06-22:
Note: I have a rewrite of BrcmPatchRAM to use IOUSBHost* instead of the legacy interfaces.  It may be useful at some point, so I'll try to finish it and add as separate branch.

sebinouse, 2015-06-22:
Good to see the progress here ! Let us know when you need testers ...

RehabMan, 2015-06-22:
You can build BrcmBluetoothInjector.kext from the repo right now if you wish to try it.

Don't forget to remove BrcmPatchRAM.kext.

theracermaster, 2015-06-23:
I can confirm BrcmBluetoothInjector works fine on 10.11. I just had to add the following for my USB dongle (Insignia Bluetooth 4.0):
```plist
<key>19ff_0239</key>
<dict>
	<key>CFBundleIdentifier</key>
	<string>com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport</string>
	<key>IOClass</key>
	<string>BroadcomBluetoothHostControllerUSBTransport</string>
	<key>IOProviderClass</key>
	<string>IOUSBHostDevice</string>
	<key>idProduct</key>
	<integer>569</integer>
	<key>idVendor</key>
	<integer>6655</integer>
</dict>
```

RehabMan, 2015-06-24:
Thanks for the feedback.  I saw your pull request on the-darkvoid's repo.  I can try to pull from here too before I push the next build.

RehabMan, 2015-06-24:
Merge went sort of ok.  Please check it.

theracermaster, 2015-06-24:
@RehabMan,
Cloned and built, still working fine.

sebinouse, 2015-06-26:
The Injector is OK for me ! Thanks

AzureWave AW-CE123H with BCM94352HMB, Bluetooth 20702A0 (13d3 3404)

RehabMan, 2015-06-27:
@sebinouse ... thanks for the feedback.

RehabMan, 2015-08-21:
It is interesting... BrcmPatchRAM.kext actually works if you kextload it instead of installing to /S/L/E.

eg.

```
sudo mkdir /kexts
sudo cp -R ./build/Products/Debug/BrcmPatchRAM.kext /kexts
sudo kextutil /kexts/BrcmPatchRAM.kext
```

It will load firmware, register the native bt personality, and bt kexts will load.

But it doesn't work if installed (bt kexts never start), and it doesn't work after sleep (there is another issue after sleep as the re-probe takes more than 400ms, but that is easily fixed).

It seems something wrong with the dynamic publishing of the bt iokit personalities...

RehabMan, 2015-09-11:
Interesting finding with master branch on 10.11:
- setConfiguration in probe makes the native *Broadcom* BT drivers fail
- setConfiguration is needed to check/load firmware
- setConfiguration when done in IOath3kfrmwr.kext for AR3011 does not cause the base BT drivers to fail

It will be interesting if the same is happening with usbhost branch.

RehabMan, 2015-09-14:
Fixed with BrcmPatchRAM2.kext.  Build available as of 2015-0913.

closed 2015-09-14 by RehabMan