RehabMan/OS-X-BrcmPatchRAM #12: BrcmPatchRAM2 crash during startup (Dell 1820A: VendorID=0A5C, ProductID=6412)

syscl, 2016-04-03:
- OS X Version: 10.11.4
- Device product ID: 6412
- Device vendor ID: 0A5C
- BrcmPatchRAM2 version used: 2.2.3

Both BrcmPatchRAM2.kext and BrcmFirmwareData.kext injected by Clover injection. 

BrcmPatchRAM2 shows endless loop：
```sh
Apr  3 19:53:35 lightingdeMacBook-Pro kernel[0]: BrcmPatchRAM2: [0a5c:6412]: device request failed ("0xe00002ed (UNDEFINED)" 0xe00002ed).
Apr  3 19:53:35 lightingdeMacBook-Pro kernel[0]: BrcmPatchRAM2: [0a5c:6412]: Not responding - Delaying next read.
```

I cannot boot into the OS X.  Sometimes the BrcmPatchRAM2.kext & BrcmFirmwareData.kext work to power up the bluetooth, but most of the time it fails.

Any other information have to be submitted?

RehabMan, 2016-04-04:
You probably have Android File Transfer installed.

RehabMan, 2016-04-04:
Note: preferred/recommended installation is BrcmPatchRAM2+BrcmFirmwareRepo in /S/L/E or /L/E.

syscl, 2016-04-04:
I do not have Android File Transfer. 

Will try to install BrcmPatchRAM2+BrcmFirmwareRepo in /S/L/E or /L/E, but I think the location may not cause the issue mention above. Any other information have to upload for a future explore?

RehabMan, 2016-04-04:
Not possible to provide any diagnosis without hardware details/ioreg/etc.
Use the forums.

You must install in the recommended way.

closed 2016-04-15 by RehabMan

onejay09, 2018-02-28:
im having the same issue, it sometimes fails to boot

sierra 10.12.6
latest clover
skylake 100 series

[Archive.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/1769029/Archive.zip)

RehabMan, 2018-03-05:
Use the forums.  This is not the appropriate place to troubleshoot your mistake.

onejay09, 2018-03-08:
wow such arrogance:
2018-02-28 21:44:53.476823+0000 0x104      Default     0x0                  0      kernel: (BrcmPatchRAM2) BrcmPatchRAM2: [0489:e07a]: Not responding - Delaying next read.
2018-02-28 21:44:53.487660+0000 0x35b      Default     0x0                  0      kernel: (BrcmPatchRAM2) BrcmPatchRAM2: [0489:e07a]: device request failed ("0xe00002d8 (UNDEFINED)" 0xe00002d8).
2018-02-28 21:44:53.499370+0000 0x35b      Default     0x0                  0      kernel: (BrcmPatchRAM2) BrcmPatchRAM2: [0489:e07a]: continuousRead - Failed to queue read (0xe00002d8)
2018-02-28 21:45:18.374648+0000 0x11f      Default     0x0                  0      kernel: (IOBluetoothFamily) **** [IOBluetoothFamily][SearchForTransportEventTimeOutHandler] -- Missing Bluetooth Controller Transport!
2018-02-28 21:45:48.603018+0000 0x3e7      Default     0x0                  0      kernel: busy timeout[0], (60s), kextd wait(0): 'AppleUSBHostLegacyClient', 'BrcmPatchRAM2', 'IOUSBHostInterface', 'IOUSBHostInterface', 'IOUSBHostInterface', 'IOUSBHostInterface'
2018-02-28 21:46:48.616231+0000 0x3e7      Default     0x0                  0      kernel: busy timeout[1], (60s), kextd wait(0): 'AppleUSBHostLegacyClient', 'BrcmPatchRAM2', 'IOUSBHostInterface', 'IOUSBHostInterface', 'IOUSBHostInterface', 'IOUSBHostInterface'
2018-02-28 21:46:48.681334+0000 0x3e7      Default     0x0                  0      kernel: IOConsoleUsers: time(0) 0->0, lin 0, llk 1,

this looks like you never fixed this issue, im providing what this OP never did.

RehabMan, 2018-03-08:
Use the forums, so you can provide complete "Problem Reporting" files.

Read FAQ, "Problem Reporting" again.  Carefully.  Attach all requested files/output.
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/
Use the gen_debug.sh tool mentioned in the FAQ, that way it is less likely you'll omit something.

In case it is not clear, please DO NOT reply here.

RehabMan, 2018-03-08:
Also, anyone with this problem should be looking into changing the IOSleep timings such as what tluck did in the tluck fork.

I asked tluck to narrow down the changes to what were actually required, but that work was never done, so I haven't bothered to address this issue any further (community work requires cooperation from the community).

For an example of addressing this issue (and some of the issues it may encompass), read here:
https://www.tonymacx86.com/threads/help-usb-devices-bluetooth-wifi-stop-working-after-sleep.246218/