RehabMan/OS-X-BrcmPatchRAM #13: Qualcomm Atheros, AR9285 Wireless Network Adapter (PCI-Express)

JahWorks, 2016-04-15:
Hi pike. I  use clover and I have one Atheros Vendor 168C, DEV 002B, SUBVENDOR 106B, SUBDEV 008F and i installed the 2 kexts:

BrcmPatchRAM2 and BrcmFirmwareData in clover EFI kexts but i cant sucess with the bluetooth.

And funny that already installed it in the version of 10.11.3 but not tested:

http://cl.ly/fiZe/bluetooth%20works%2010.11.jpg

I att my ioreg 
[image]
[G75vw bios 202.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/221747/G75vw.bios.202.zip)

RehabMan, 2016-04-15:
This driver is for Broadcom BT, not Atheros.

BTW, I don't answer to 'Pike'.

closed 2016-04-15 by RehabMan

JahWorks, 2016-04-28:
Sorry! it was a mistake, RehabMan.
The driver is Broadcom as you said.
The bluetooth works in pref. pane but the icon is offline
[image]
[image]
[image]