RehabMan/OS-X-BrcmPatchRAM #14: dependency 'com.no-one.BrcmFirmwareStore' not found

ylluminate, 2016-05-15:
Getting these when following the Mac OS X native (Mac Pro Mid 2010) instructions of just installing `BrcmPatchRAM2.kext` to /S/L/E/:
```
(standard input):6290:May 14 23:29:08 hostname com.apple.kextcache[4280]: BrcmPatchRAM2.kext - dependency 'com.no-one.BrcmFirmwareStore' not found.
(standard input):6291:May 14 23:29:08 hostname com.apple.kextcache[4280]: BrcmPatchRAM2.kext is missing dependencies (including anyway; dependencies may be available from elsewhere)
(standard input):6292:May 14 23:29:08 hostname com.apple.kextcache[4280]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext BrcmPatchRAM2.kext
(standard input):6312:May 14 23:29:32 hostname com.apple.kextcache[4280]: BrcmPatchRAM2.kext - dependency 'com.no-one.BrcmFirmwareStore' not found.
(standard input):6327:May 14 23:29:40 hostname com.apple.kextcache[4280]: BrcmPatchRAM2.kext - dependency 'com.no-one.BrcmFirmwareStore' not found.
(standard input):6328:May 14 23:29:40 hostname com.apple.kextcache[4280]: Prelink failed for com.no-one.BrcmPatchRAM2; omitting from prelinked kernel.
(standard input):7026:May 14 23:32:03 localhost com.apple.kextd[63]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/BrcmPatchRAM2.kext"
(standard input):7028:May 14 23:32:03 localhost com.apple.kextd[63]: /System/Library/Extensions/BrcmPatchRAM2.kext - dependency 'com.no-one.BrcmFirmwareStore' not found.
(standard input):7031:May 14 23:32:03 localhost com.apple.kextd[63]: /System/Library/Extensions/BrcmPatchRAM2.kext - dependency 'com.no-one.BrcmFirmwareStore' not found.
(standard input):7032:May 14 23:32:03 localhost com.apple.kextd[63]: Can't load /System/Library/Extensions/BrcmPatchRAM2.kext - failed to resolve dependencies.
(standard input):7034:May 14 23:32:03 localhost com.apple.kextd[63]: Load com.no-one.BrcmPatchRAM2 failed; removing personalities from kernel.
```

RehabMan, 2016-05-15:
You did not install correctly.  Must install both BrcmPatchRAM2.kext and BrcmFirmwareRepo.kext.

Full installation instructions/requirements are in the README.

closed 2016-05-15 by RehabMan

ylluminate, 2016-05-15:
Thanks for specifying; since you said to follow the instructions [here](https://github.com/the-darkvoid/BrcmPatchRAM/blob/master/README-Mac.md), and installing this kext was not specified, I thought I was to only install the singular kext.  Glad to know now.

RehabMan, 2016-05-15:
I have updated the Mac README in my fork.

ylluminate, 2016-05-15:
Great, thanks!  You may also want to relink the instructions link on the main page to your repo vs: https://github.com/robvanoostenrijk/BrcmPatchRAM/blob/master/README-Mac.md

RehabMan, 2016-05-15:
done.