RehabMan/OS-X-BrcmPatchRAM #19: BCM4352

celvdd, 2016-06-14:
Hi
I have the BCM4352 WIFI AC + BT 4.0 card installed in an Acer V5 laptop. On Windows the BT is working, but on Mac (El Capitan and Yosemite), the Bluetooth can't connect or find my devices. Sometimes my Android phone is found and connected, and one moment I could transfer a file, but my Microsoft Sculpt Comfort BT mouse can't connect (sometimes it's found). My Bluetooth earbuds and speaker are not even found. Wifi 5GHz is working perfect.

RehabMan, 2016-06-14:
What device-id?  Did you install the kexts correctly (to the system volume)?

RehabMan, 2016-06-14:
Also, you mention an Android device.  Make sure you did not install Android File Transfer.

celvdd, 2016-06-14:
How can I find the device-id?
And why does Android file transfer is a problem?

RehabMan, 2016-06-14:
You can find your id in System Information->Bluetooth or System Information->USB.

The issues with Android File Transfer are covered in the FAQ: 
http://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/

RehabMan, 2016-06-14:
If you have Android File Transfer installed, it is likely the cause of your problems...

celvdd, 2016-06-14:
I will remove AFT and see if it works. This is the usb information, I used the device id and company id for making my BT be recognized (edited BT kext).

BCM20702A0:

  Productcode:	0x3404
  Fabrikantcode:	0x13d3  (AzureWave Technologies, Inc.)
  Versie:	1.12
  Serienummer:	40E2303AE2B0
  Snelheid:	Maximaal 12 Mb/sec
  Fabrikant:	Broadcom Corp
  Locatiecode:	0x14400000 / 2
  Beschikbare stroom (mA):	1000
  Vereiste stroom (mA):	0
  Extra stroom in werking (mA):	0
  Ingebouwd:	Ja

celvdd, 2016-06-14:
Hello
I have removed AFT, but the bluetooth still won’t connect to the MS Sculpt mouse. Any other ideas?

RehabMan, 2016-06-15:
13d3:3404 is not known for good compatibility.

You can check in Windows to see if there is updated firmware, and if there is try it.

celvdd, 2016-06-15:
I installed the latest BT driver (BT wasn't working OOTB in WIndows after changing the Wifi+BT card to the BCM). Also, how would Windows affect the Mac BT driver?

RehabMan, 2016-06-15:
You can look in Device Manager, Properties (go the part that lists the device driver components). One of the components should be the HEX firmware file.  That indicates the firmware you're using on Windows.  You can compare it to the firmware used by BrcmPatchRAM.

Also, if you restart from Windows into OS X, BrcmPatchRAM will not load its firmware and you'll end up using the firmware that Windows pushed to the device.

RehabMan, 2016-06-20:
Also, no need to "edit IDs in kexts" with your BT hardware.  Just install BrcmPatchRAM kexts as detailed in the README.  If you're doing something more than that, you're doing something wrong.

closed 2016-06-20 by RehabMan

celvdd, 2016-06-21:
But my bluetooth was not recognized without editing the device ID.

RehabMan, 2016-06-21:
The device you have is already in the BrcmPatchRAM kext.

celvdd, 2016-06-21:
But it was only recognised after the edit..

RehabMan, 2016-06-21:
You must have been doing something wrong.  The 13d3:3404 has been in the kext a long time.  Many people make the mistake of installing Android File Transfer, have USB problems, or do not install the BrcmPatchRAM kexts properly.

celvdd, 2016-06-21:
okay thank you, how can i get back the original Bluetooth Controller kext?

RehabMan, 2016-06-21:
Run the OS X installer (no need to erase the target partition).  It will restore all kexts to vanilla.

celvdd, 2016-06-21:
Okay, I hope it won't break my build because I've had a lot of trouble
making all the USB devices etc. working. Also my graphics were a hard time
so I hope I can restore everything. Thank you for the help.

RehabMan, 2016-06-21:
If you do your installation correctly, you end up having no patched kexts in the system volume...  only additional kexts, and all patches via Clover config.plist.

celvdd, 2016-06-21:
But I have HD4400 graphics with need to be patched, as well as the elan
track pad driver for multi gestures, as well as the patch for enabling my
USB's (track pad, BT, touchscreen, ...)

RehabMan, 2016-06-21:
No need to patch the Azul graphics kexts on disk for HD4400.  All can be done from config.plist.

As far as ELAN PS2 driver... it is not a "patched kext".  It is an additional kext and will not be deleted by the OS X installer.

No patching necessary for USB either...  All can be done with additional kexts and/or ACPI.

celvdd, 2016-06-21:
The graphics are patched through clover, but I needed to load an extra
hd5000 next for it to work, the USB is only working when IOUSBHostefamily
is patched.

celvdd, 2016-06-21:
But I will try the reinstall tonight, but I don't think it will work
because BT wasn't working on Yosemite either (only finding my phone, no
mouse or headset)

cel.van.den.daele@gmail.com>:

RehabMan, 2016-06-21:
BT requires compatible hardware and the software that goes with it.

celvdd, 2016-06-21:
Okay I will try it with the original kexts. Do you know a reason why BT
seems to be working, but doesn't connect to my mouse, headset etc?

RehabMan, 2016-06-21:
No idea without details... eg. which specific BT controller you have, and whether you installed BrcmPatchRAM correctly... no Android File Transfer, etc.

celvdd, 2016-06-21:
It's the bcm20702a0, I tried without AFT, and the BrcmPatchRam2  and the
Repo are both installed in S/L/E
I also tried them through Clover injection (with Data)

RehabMan, 2016-06-21:
Make sure you check system.log to see if firmware is loading.  And check System Information->Bluetooth to see the firmware version is as expected.

celvdd, 2016-06-21:
This is the loaded firmware: 4.4.5f3 17904

RehabMan, 2016-06-21:
That is not a bluetooth firmware controller version number.  You're looking in the wrong place.

celvdd, 2016-06-21:
I’ve reinstalled El Capitan, BT is still not finding devices, although my android phone found my pc.
This is the system.log output:
Jun 21 23:04:21 MacBook-Pro-van-Cel com.apple.kextd[45]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/BrcmPatchRAM2.kext"
Jun 21 23:04:22 MacBook-Pro-van-Cel com.apple.kextd[45]: kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/BrcmPatchRAM2.kext"
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: Version 2.2.3 starting on OS X Darwin 15.5.
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmFirmwareStore.
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmPatchRAMResidency.
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM: Loaded firmware "BCM20702A1_001.002.014.1443.1479_v5575.zhx" from resources.
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM: Decompressed firmware (29713 bytes --> 70105 bytes).
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM: Firmware is valid IntelHex firmware.
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: [13d3:3404]: USB [40E2303AE2B0 v274] "BCM20702A0" by "Broadcom Corp"
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: [13d3:3404]: Firmware upgrade completed successfully.
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: [13d3:3404]: Published new IOKit personality.
Jun 21 23:04:22 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: Processing time 0.534 seconds.
Jun 21 23:04:25 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: Version 2.2.3 starting on OS X Darwin 15.5.
Jun 21 23:04:25 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: [13d3:3404]: USB [40E2303AE2B0 v274] "BCM20702A0" by "Broadcom Corp"
Jun 21 23:04:25 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: [13d3:3404]: Firmware upgrade not needed.
Jun 21 23:04:25 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: [13d3:3404]: Found existing IOKit personality "com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport".
Jun 21 23:04:25 MacBook-Pro-van-Cel kernel[0]: BrcmPatchRAM2: Processing time 0.13 seconds.

celvdd, 2016-06-21:
And this is the Firmware Version: v14 c5575

RehabMan, 2016-06-21:
That is the latest/correct firmware for 13d3:3404.

celvdd, 2016-06-22:
It still doesn't find devices

RehabMan, 2016-06-22:
You need to use the forums.  You probably have an antenna issue or some other environmental or compatibility issue.

This is not the place for troubleshooting.