RehabMan/OS-X-BrcmPatchRAM #20: No Bluetooth Devices Shown

Reservations, 2016-06-24:
Device product ID: 0x0a2b
Device vendor ID: 0x8087
BrcmPatchRAM version used: RehabMan-BrcmPatchRAM-2016-0621
Dump of BrcmPatchRAM debug output from /var/log/system.log showing the firmware upload failure:
`Jun 23 22:40:13 localhost kernel[0]: BrcmPatchRAM2: Version 2.2.6 starting on OS X Darwin 15.5.
Jun 23 22:40:13 localhost kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmFirmwareStore.
Jun 23 22:40:13 localhost kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmPatchRAMResidency.
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM: Loaded firmware "BCM20702A1_001.002.014.1502.1764_v5860.zhx" from resources.
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM: Decompressed firmware (30360 bytes --> 71148 bytes).
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM: Firmware is valid IntelHex firmware.
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: USB [5CF370745ABB v274] "BCM920702 Bluetooth 4.0" by "Broadcom Corp"
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: Firmware upgrade not needed.
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: Found existing IOKit personality "com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport".
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: Processing time 0.704 seconds.
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: Version 2.2.6 starting on OS X Darwin 15.5.
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: uploadFirmware could not open the device!
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: Found existing IOKit personality "com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport".
Jun 23 22:40:14 localhost kernel[0]: BrcmPatchRAM2: Processing time 0.2 seconds.
Jun 23 22:47:38 localhost kernel[0]: BrcmPatchRAM2: Version 2.2.6 starting on OS X Darwin 15.5.
Jun 23 22:47:38 localhost kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmFirmwareStore.
Jun 23 22:47:38 localhost kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmPatchRAMResidency.
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM: Loaded firmware "BCM20702A1_001.002.014.1502.1764_v5860.zhx" from resources.
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM: Decompressed firmware (30360 bytes --> 71148 bytes).
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM: Firmware is valid IntelHex firmware.
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: USB [5CF370745ABB v274] "BCM920702 Bluetooth 4.0" by "Broadcom Corp"
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: Firmware upgrade not needed.
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: Found existing IOKit personality "com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport".
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: Processing time 2.204 seconds.
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: Version 2.2.6 starting on OS X Darwin 15.5.
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: uploadFirmware could not open the device!
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: [0a5c:21e8]: Found existing IOKit personality "com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport".
Jun 23 22:47:40 Khas-iMac kernel[0]: BrcmPatchRAM2: Processing time 0.2 seconds.`

RehabMan, 2016-06-24:
8087:0a2b is Intel,not Broadcom.
Make sure you don't have Android File Transfer installed.

RehabMan, 2016-06-24:
This thread should be used for troubleshooting:
http://www.tonymacx86.com/threads/brcmpatchram-upload-firmware-into-broadcom-bluetooth-usb-devices.150661/

RehabMan, 2016-07-05:
Closing issue due to no further response.

closed 2016-07-05 by RehabMan