RehabMan/OS-X-BrcmPatchRAM #22: Same problem as in "BrcmPatchRAM2 fails when in Clover/kexts"?

mklengel, 2016-07-07:
Hi RehabMan,

first I would like to thank you for BrcmPatchRAM. Bluetooth on my Asus G750JM seems to work but I detected some a message in /var/log/system. The firmware is updated (no "4096").

# cat /var/log/system.log | grep -i brcm[fp]
Jul  7 17:33:34 localhost kernel[0]: BrcmPatchRAM2: Version 2.2.7 starting on OS X Darwin 15.5.
Jul  7 17:33:34 localhost kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmFirmwareStore.
Jul  7 17:33:34 localhost kernel[0]: BrcmPatchRAM2: Published new IOKit personality for BrcmPatchRAMResidency.
**Jul  7 17:33:37 Michaels-MacBook-Pro com.apple.kextd[46]: Can't open CFBundle for /EFI\CLOVER\kexts\10.11\BrcmFirmwareData.kext.
Jul  7 17:33:37 Michaels-MacBook-Pro com.apple.kextd[46]: Can't open CFBundle for /EFI\CLOVER\kexts\10.11\BrcmFirmwareData.kext.
Jul  7 17:33:37 Michaels-MacBook-Pro com.apple.kextd[46]: Can't open CFBundle for /EFI\CLOVER\kexts\10.11\BrcmFirmwareData.kext.**
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM: Loaded compressed embedded firmware for key "BCM20702A1_001.002.014.1443.1479_v5575".
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM: Decompressed firmware (29713 bytes --> 70105 bytes).
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM: Firmware is valid IntelHex firmware.
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: [13d3:3404]: USB [54271EBE0868 v274] "BCM20702A0" by "Broadcom Corp"
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: [13d3:3404]: Firmware upgrade not needed.
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: [13d3:3404]: Published new IOKit personality.
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: Processing time 4.598 seconds.
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: Version 2.2.7 starting on OS X Darwin 15.5.
Jul  7 17:33:37 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: [13d3:3404]: USB [54271EBE0868 v274] "BCM20702A0" by "Broadcom Corp"
Jul  7 17:33:38 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: [13d3:3404]: Firmware upgrade not needed.
Jul  7 17:33:38 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: [13d3:3404]: Found existing IOKit personality "com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport".
Jul  7 17:33:38 Michaels-MacBook-Pro kernel[0]: BrcmPatchRAM2: Processing time 0.65 seconds.

RehabMan, 2016-07-07:
You installed incorrectly.  All kexts should be installed to the system volume.  Please read the README.

closed 2016-07-07 by RehabMan

mklengel, 2016-07-07:
Did I misunderstand your README (Section INSTALLATION)?

**BrcmFirmwareData.kext: Most appropriate for EFI/Clover/kexts. BrcmFirmwareRepo.kext, while much more memory efficient, cannot be injected as can BrcmFirmwareData.kext**

RehabMan, 2016-07-08:
Requires tweaking the Info.plist a bit due to the dynamic loading from BrcmPatchRAM.
Change IOProviderClass from disabled_IOResources to IOResources if you want to try it.

But it is not really supported... the BrcmFirmwareData.kext has grown and is really far too large/inefficient to be injected.

The recommended kexts are BrcmPatchRAM[2].kext + BrcmFirmwareRepo.kext installed to the system volume (where all kexts should be installed).

mklengel, 2016-07-08:
No problem. I tend to customize everything in Clover (Patching, Kexts, Plists) but I'm sure you are right.

Thank you for you support
Michael