RehabMan/OS-X-BrcmPatchRAM #28: Handoff DW1830 (​BCM943602BAED)

seekerPL, 2017-03-27:
I ordered BCM943602BAED and replace my BCM94352Z on Dell e7450.
I decided to replace it due to fact BCM94352Z was not properly working with Handoff (t was working from mac to iPhone, but not opposite way, also I had some problems with Universal Clipboard and auto unlocking function from Apple Watch).

DW1830@BCM943602BAED
IDs:
wireless = 14e4:43ba
bluetooth = 0a5c:6410

kexts installed:

BrcmFirmwareRepo.kext
BrcmPatchRAM2.kext
FakePCIID.kext

All kext to patches in Cover config were removed. They were previously added for BCM94352Z.

To let you know, currently I do not see any reason tor replace BCM94352Z with BCM943602BAED. Yes, it require less patching, only BrcmFirmwareRepo.kext and BrcmPatchRAM2.kext for Bluetooth Wifi works OOB, but you will get same problems with functionality.

Airdrop - works both ways
Continuity - partially working
Universal Clipboard - not working
Apple Watch auto-unlock not working
Hotspot - working
find My iPhone works until restart, then it is disabled again?!? why?

e7450 updated to latest 10.12.3 from 10.12.1

[config.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/871875/config.zip)

RehabMan, 2017-03-29:
This is a problem with device/firmware compatibility.
No issue with this kext (it just loads firmware as Windows would).

closed 2017-03-29 by RehabMan

johnyoonh, 2017-06-16:
@seekerPL I saw your thread from: http://forum.osxlatitude.com/index.php?/topic/9670-dw1830-bcm943602baed/. How were you able to get your wifi working? I've been stuck with this problem for weeks

seekerPL, 2017-11-10:
DW1830  Wifi should work OOB. Probably you have to patch your DSDT.