RehabMan/OS-X-BrcmPatchRAM #29: BCM43224 Wi-fi keeps disconnecting over and over when BrcmPatchRAM2 is loaded

RR7, 2017-04-30:
it doesn't matter if I put it in clover or /l/e or /s/l/e, it always does this:

wifi keeps going off then on, you can see it doing in in the networks preferences. if I remove the patchram2 kext then all is well.

the wifi is a BCM43224
the BT is BCM20702A0

not sure why this kext would keep toggling the wifi on and off, happy to try any suggestions if you have any.

RehabMan, 2017-05-01:
This is not the appropriate place to troubleshoot your issue.
Use the forums.

closed 2017-05-01 by RehabMan

RR7, 2017-05-03:
thanks, I posted here:

http://www.insanelymac.com/forum/topic/323346-brcmpatchram2-keeps-resetting-wi-fi-on-and-off-mostly-off/

hope thats the right place.