RehabMan/OS-X-BrcmPatchRAM #35: Help,BCM943602cs Not Work, support? 0A5C:21FF

andyhuai, 2017-09-07:
Hey, my card is BCM943602cs PCI-E, now WiFi is OK.
but bluetooth did not work fine.
I found that, if I start windows 10 ,and turn-on BT ,then restart to MAC,
the BT work! If power off the power, BT is gone.

Another, the BT work time, the idProducr is 0x8290, else is 0x21FF,
this is so puzz me!
can I use this kext to fix my BT ??

thanks!

RehabMan, 2017-09-29:
There is no support for firmware load on your device in this kext.
You might check in Windows to see which .hex is being used.
You might be able to build an injector kext with that file (see README, look at example injectors).

closed 2017-09-29 by RehabMan

andrewclink, 2018-05-28:
This is pretty old but I keep finding this issue when searching for a solution.

My card is supposedly a DW1830 that initially starts as 0A5C:21FF. Through trial and error I finally found a way to get it working.

No Windows 10 drivers that I could find match `21ff`, but if you choose to "update driver" and then "manually" and "from a list of drivers" you will have the option to force a different driver to attach. Choose Bluetooth -> DW1830 (or whatever yours is supposed to be). The process appears to try to work, but ultimately fails when the device reenumerates with the new PID.

Mine shows up with a PID of 0A5C:6410 which BrcmPatchRAM2 then handles the next boot.

I extracted the firmware (BCM20703A1_001.001.005.0214.0403.hex) and attempted to create a new key in BrcmFirmwareRepo but I'm unable to test because I don't know what causes the PID to flip back to 21ff. If I catch it working I'll create a PR with my changes, as I've seen this issue go unsolved in several threads on different forums.