RehabMan/OS-X-BrcmPatchRAM #40: Bluetooth on the module BCM943602BAED DW1830

walkman8196, 2017-10-08:
[MacBook Pro — Константин.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/1366752/MacBook.Pro.zip)
Hello RehabMan please add support Bluetooth: BCM20703A1  0a5c:6410
Although the resources present BCM20703A1 firmware / BrcmPatchRAM does not load it.
I put  io registry

walkman8196, 2017-10-08:
[New_Bt_firmware_BCM20703A1.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/1366756/New_Bt_firmware_BCM20703A1.zip)

RehabMan, 2017-10-09:
Support for 0a5c:6410 already present.
Firmwares are updated automatically from Windows drivers via firmware.rb.
Last/latest drivers I know of on station-drivers: http://www.station-drivers.com/index.php?option=com_remository&Itemid=352&func=fileinfo&id=2130&lang=en

closed 2017-10-09 by RehabMan

walkman8196, 2017-10-09:
Strangely, since the 5th time the firmware has booted, but the bluetooth does not work properly and the bluetooth does not see the device

RehabMan, 2017-10-09:
Use the forums.