RehabMan/OS-X-BrcmPatchRAM #43: Failed to boot into RecoveryHD or macOS Installer

wsy, 2017-10-30:
I'm using Lenovo K29, running macOS High Sierra with BCM94352 card.
I cannot boot into RecoveryHD(10.13) and nor can I boot into macOS High Sierra Installer.
The last screen in verbose boot log keeps flushing these words:

> BrcmPatchRAM: Loaded compressed embedded firmware for key "BCM20702A1_001.002.014.1443.1493_v5589".
> BrcmPatchRAM: Decompressed firmware (28711 bytes --> 70105 bytes).
> BrcmPatchRAM: Firmware is valid IntelHex firmware.
> BrcmPatchRAM2: [04ca:200b]: USB [3010B3A2F4FA v274] "BCM20702A0" by "Broadcom Corp"
> BrcmPatchRAM2: [04ca:200b]: Firmware upgrade not needed.
> BrcmPatchRAM2: [04ca:200b]: Published new IOKit personality.
> BrcmPatchRAM2: Processing time 122.465 seconds.
> BrcmPatchRAM2: Version 2.2.7 starting on OS X Darwin 17.0.
> BrcmPatchRAM2: [04ca:200b]: USB [3010B3A2F4FA v274] "BCM20702A0" by "Broadcom Corp"
> BrcmPatchRAM2: [04ca:200b]: Firmware upgrade not needed.
> BrcmPatchRAM2: [04ca:200b]: Published new IOKit personality.
> BrcmPatchRAM2: Processing time 0.34 seconds.
> BrcmPatchRAM2: Version 2.2.7 starting on OS X Darwin 17.0.
> BrcmPatchRAM2: [04ca:200b]: USB [3010B3A2F4FA v274] "BCM20702A0" by "Broadcom Corp"
> BrcmPatchRAM2: [04ca:200b]: Firmware upgrade not needed.
> BrcmPatchRAM2: [04ca:200b]: Published new IOKit personality.
> BrcmPatchRAM2: Processing time 0.32 seconds.
> BrcmPatchRAM2: Version 2.2.7 starting on OS X Darwin 17.0.
> BrcmPatchRAM2: [04ca:200b]: USB [3010B3A2F4FA v274] "BCM20702A0" by "Broadcom Corp"
> BrcmPatchRAM2: [04ca:200b]: Firmware upgrade not needed.
> BrcmPatchRAM2: [04ca:200b]: Published new IOKit personality.
> BrcmPatchRAM2: Processing time 0.29 seconds.
> BrcmPatchRAM2: Version 2.2.7 starting on OS X Darwin 17.0.
> BrcmPatchRAM2: [04ca:200b]: USB [3010B3A2F4FA v274] "BCM20702A0" by "Broadcom Corp"
> BrcmPatchRAM2: [04ca:200b]: Firmware upgrade not needed.
> BrcmPatchRAM2: [04ca:200b]: Published new IOKit personality.
> BrcmPatchRAM2: Processing time 0.29 seconds.
> BrcmPatchRAM2: Version 2.2.7 starting on OS X Darwin 17.0.
> ......(**These five lines loops forever**)

On the other hand, **booting into macOS is OK**. This problem only occurs on RecoveryHD and macOS Installer.

Here's my brief configuration:(Detailed configurations are in the attachment)
[ConfigurationAndClover.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/1426318/ConfigurationAndClover.zip)

i5 3210M
HD4000 ig-platform-id 0x01660003
Wifi: 14e4:43b1
Bluetooth: 04ca:200b (BCM20702a0)

Clover version 4259
BrcmNonPatchRAM2.kext
BrcmPatchRAM2.kext
BrcmFirmwareData.kext

RehabMan, 2017-10-30:
You don't need BrcmPatchRAM kexts for the recovery partition.
It is not an "essential" kext, so no need to place it in EFI/Clover/kexts/Other.
Suggest you remove it and leave Clover/kexts only for essential kexts.

Your core problem, however, is likely USB related.
Use the forums if you need to troubleshoot.

closed 2017-10-30 by RehabMan

wsy, 2017-10-31:
It's ok that you don't want to add an "if-else" statement determining whether it's a normal macOS boot or recovery environment.
I still wonder why recovery partition does not need these kexts to get the bluetooth working?

At last, thank you very much for your great work! Many of the kexts I'm using are from you! Thank you!

RehabMan, 2017-10-31:
It is not clear that your issue is related to this kext.
Looks like a USB configuration issue to me.
Like I wrote earlier, use the forums.

RehabMan, 2017-11-08:
The recovery environment may need BrcmPatchRAM kexts to get BT working.
But BT is not really needed for the recovery environment as the kinds of things you do in recovery do not usually involve BT.

wmchris, 2017-12-12:
There are use cases where bt is needed, so this bug is a problem

RehabMan, 2017-12-12:
Make sure your USB configuration is correct.
Read here:
https://www.tonymacx86.com/threads/guide-creating-a-custom-ssdt-for-usbinjectall-kext.211311/
https://www.tonymacx86.com/threads/guide-usb-power-property-injection-for-sierra-and-later.222266/
https://www.tonymacx86.com/threads/guide-10-11-usb-changes-and-solutions.173616/

If you wish to address the issue/troubleshoot, use the forums.

wmchris, 2017-12-12:
i cant use the forums (banned for mentioning a tool from another webpage in a self written tutorial).
the usb configuration is correct, to be honest: its not even an usb device - it just randomly appeared on a full working system after 3 month of usage.

RehabMan, 2017-12-13:
No idea what you mean by "its not even an usb device".
The bluetooth controller is always connected via USB.
You will need to use the forums.  This is not the proper area to troubleshoot your issue.

wmchris, 2017-12-13:
i cant use a forum where i'm banned.

this device is a pcie combo card. so its technically using usb.
all required data are at https://github.com/wmchris/DellXPS15-9550-OSX - as you can see it doesnt require usbinjectall. its completly ssdt based.
this issue only occurs after triggering the osx updater or installer, and even then not everytime.

RehabMan, 2017-12-14:
> as you can see it doesnt require usbinjectall. its completly ssdt based.

Nonsense.  SSDT-UIAC-9550.aml does nothing without USBInjectAll.kext.

wmchris, 2017-12-14:
thanks. forgot to delete this file

RehabMan, 2017-12-14:
Without USBInjectAll.kext and a properly created SSDT for it (UIAC.RMCF), your USB configuration is likely wrong.

wmchris, 2017-12-16:
there are multiple usb ssdts in my repository which configure usb, even usb-c with hotplug. working with this setup for 2 years now without any problem. first issue was the problem reported by OP, and only on a recovery kernel for 10.13.2, 10.13.1 works w/o an issue.

RehabMan, 2017-12-16:
See FAQ for "Problem Reporting" requirements.
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/

wmchris, 2017-12-17:
this is a bug report. not a support request on a commercial forum. i will help you and the original developer if you need some testing or if something is unclear, but then you have to tell me what you need.

tbh: this is a loop. maybe a simple loop counter would fix this issue.

RehabMan, 2017-12-17:
Please submit your problem reporting files so I can see your actual configuration.

wmchris, 2017-12-17:
i uploaded all files on github - as part of the tutorial. because this bug occurs at the upgrade kernel, i cannot supply the output of kextstat.

RehabMan, 2017-12-17:
I need the files requested in FAQ, "Problem Reporting".
I cannot predict what the final setup might be from your github content.

FYI: I was not able to reproduce the reported bug on my Lenovo u430.  I have a correct USB configuration...
- USBInjectAll.kext in EFI/Clover/kexts/Other
- SSDT content to configure the ports/USBInjectAll correctly (# of ports within the 15-port limit, unused ports eliminated, correct port-count for the controller, correct UsbConnector values)
- BrcmPatchRAM2.kext + BrcmFirmwareData.kext in Clover/kexts/Other (along with other "essential" kexts.

There was no such problem as mentioned at the beginning of this issue.  And I verified that native Broadcom bluetooth is started by looking at ioreg (via IORegistryExplorer v2.1).   Verbose logs looked good at startup.  Tried both warm boot (firmware already loaded) and cold boot (BrcmPatchRAM had to load firmware).

So... yes... I suspect your issue is an incorrect USB configuration.

RehabMan, 2017-12-18:
FYI...
I don't see ioreg (from normal boot), patchmatic -extract (also from normal boot), nor finalized EFI/Clover on your github (all needed to clearly see what you are actually doing).
And... I also don't see anything in the setup there that will configure USB correctly... you have a UIAC.RMCF, but USBInjectAll.kext seems to be (likely) missing from EFI/Clover/kexts/Other.