RehabMan/OS-X-BrcmPatchRAM #47: [0b05:1825] ASUSTek Computer, Inc. Qualcomm Atheros QCA61x4A Bluetooth 4.1 Works

oblitum, 2018-01-14:
See https://github.com/the-darkvoid/BrcmPatchRAM/issues/41. I'm duplicating it here because it's a bit confusing how both projects are evolving in different repositories.

RehabMan, 2018-01-16:
duplicating the content from that issue:

I have an ASUS ROG MAXIMUS IX FORMULA motherboard and the kexts work for causing the on-board Qualcomm Atheros QCA61x4A to recognize my Magic Mouse. It's working on macOS High Sierra 10.13.2. I'm just reporting so that it becomes listed as tested. Before the installation the Bluetooth controller was reported as functional by macOS but it couldn't scan/pair any device.

I've just put BrcmPatchRAM2.kext and BrcmFirmwareData.kext on /EFI/CLOVER/kexts/Other because I prefer to leave these extra Hackintosh kexts apart from the OS partition if possible, but I notice that the README highlights that BrcmFirmwareRepo.kext on /System/Library/Extensions is the preferred method and it's much less memory hungry, so..., is it that much better? I have 16GB RAM and I didn't get how a simple kext can be that memory hungry to my system, or is it about the controller's memory? Is there a more detailed drawback of sticking to BrcmFirmwareData.kext since I can just put it on Clover besides BrcmPatchRAM2.kext?

RehabMan, 2018-01-16:
BrcmPatchRAM2.kext will not match against 0b05:1825, as it is not in the Info.plist.
Installing those kexts will have no effect.
If you think it is having an effect on your system, perhaps you are mistaken regarding the vendor/product-id.

BrcmFirmwareData.kext uses more memory as it includes all firmwares for all supported devices instead of loading from disk on demand as BrcmFirmwareRepo.kext does.

oblitum, 2018-01-16:
OK thanks for the info. So then, I may have not checked it properly without the kexts, I will try again and see.

oblitum, 2018-01-16:
I just confirmed, tried multiple boots without the kexts, and with just `BrcmPatchRAM2.kext`, bluetooth won't recognize any devices trying to pair. Once I put the two kexts back in Clover and reboot, bluetooth readily recognizes my magic mouse and I can pair it. Whatever the two kexts are doing in my system, it's what's making my bluetooth to work.

oblitum, 2018-01-16:
Regarding the vendor:product tuple, I get it from my main partition running linux:

```
~ ❯❯❯ lsusb | grep Bluetooth
Bus 001 Device 008: ID 0b05:1825 ASUSTek Computer, Inc. Qualcomm Bluetooth 4.1=
```

oblitum, 2018-01-16:
My suspicion is that the specific product id while not being listed in the `Info.plist`, is still [serving to provide a Probe Score](https://developer.apple.com/library/content/qa/qa1076/_index.html) to load a best driver/firmware that happens to still work on this device.

RehabMan, 2018-01-18:
A kext which has no match for a certain device will have no effect.
You should use the forums and provide complete problem reporting data.

For example on tmx:
Read FAQ, "Problem Reporting" again.  Carefully.  Attach all requested files/output.
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/
Use the tool mentioned in the FAQ, that way it is less likely you'll omit something.

oblitum, 2018-01-18:
There are several matches in the [probe score table samples](https://developer.apple.com/library/content/qa/qa1076/_index.html) without product id. I don't know how this works in detal as macOS isn't my main platform, and your assertions seem to conflict with official docs. Anyway. Putting these two kexts in place is, without a doubt, what brings my Bluetooth to life, so thank you and @the-darkvoid for it. Just wanted to report and leave it as that without further digging.

closed 2018-01-18 by oblitum

RehabMan, 2018-01-18:
The fact remains that BrcmPatchRAM kexts have no support for 0b05:1825.
A kext cannot have an effect without an IOKitPersonality match for the device in question.
Period.

You have something else going on.

chwzr, 2018-03-12:
I have the same chip (QCA61x4A) onboard, and  initially it was not working on a clean install. But: I plugged in my IOGEAR GBU521 Bluetooth USB Dongle, and then unplugged it.. since then the internal bluetooth worked just fine 👯 

The IOGEAR has following hardware: BCM20702A Chipset, productID 8680, vendorID 2652, firmware version v14 c4096, HCI version 6, HCI revision 4096, LMP version 6, LMP subversion 8718

Just wondering bc its not a Atheros inside the dongle...
Thats on a Acer Swift 7 Laptop (713-51)

btw:
I know its not the right place to ask,  but I hope someone here maybe knows something: is there a chance to get Wifi working with QCA61x4A ?

Kind regards