RehabMan/OS-X-BrcmPatchRAM #54: Xcode Fails to Build - 'IOKit/usb/IOUSBDevice.h' file not found

peterlobster, 2018-09-07:
I'm trying to build from source but I keep running into this issue that prevents me from building. I would appreciate any help...

[image]

RehabMan, 2018-09-14:
Must use proper SDK depending on which target you're trying to build.

closed 2018-09-14 by RehabMan

RehabMan, 2018-09-18:
As per checked in project settings...
BrcmPatchRAM is built with SDK for 10.6.
BrcmPatchRAM2 is built with SDK for 10.11.

Note that I have all SDKs installed to my Xcode:
```
NUC6i7KYK:~ rehabman$ ls -l /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs
total 64
drwxr-xr-x  5 root  wheel  170 Jun 15 07:36 MacOSX.sdk
lrwxr-xr-x  1 root  wheel   40 Sep 14 07:22 MacOSX10.10.sdk -> /Applications/LegacySDKs/MacOSX10.10.sdk
lrwxr-xr-x  1 root  wheel   40 Sep 14 07:22 MacOSX10.11.sdk -> /Applications/LegacySDKs/MacOSX10.11.sdk
lrwxr-xr-x  1 root  wheel   40 Sep 14 07:22 MacOSX10.12.sdk -> /Applications/LegacySDKs/MacOSX10.12.sdk
lrwxr-xr-x  1 root  wheel   10 May 12 15:46 MacOSX10.13.sdk -> MacOSX.sdk
lrwxr-xr-x  1 root  wheel   39 Sep 14 07:22 MacOSX10.6.sdk -> /Applications/LegacySDKs/MacOSX10.6.sdk
lrwxr-xr-x  1 root  wheel   39 Sep 14 07:22 MacOSX10.7.sdk -> /Applications/LegacySDKs/MacOSX10.7.sdk
lrwxr-xr-x  1 root  wheel   39 Sep 14 07:22 MacOSX10.8.sdk -> /Applications/LegacySDKs/MacOSX10.8.sdk
lrwxr-xr-x  1 root  wheel   39 Sep 14 07:22 MacOSX10.9.sdk -> /Applications/LegacySDKs/MacOSX10.9.sdk
```

All legacy SDKs (collected from previous versions of Xcode) I have copied to /Applications/LegacySDKs.

I use this legacysdks.sh script to create the symlinks after setting up current Xcode:
```
#!/bin/bash

osxplatform=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform
sdkrepo=$osxplatform/Developer/SDKs/
infoplist=$osxplatform/Info.plist

for sdk in /Applications/LegacySDKs/*.sdk; do
   sudo ln -s $sdk $sdkrepo`basename $sdk`
done
ls -l $sdkrepo

sudo /usr/libexec/PlistBuddy -c "Set :MinimumSDKVersion 10.6" $infoplist
```

You cannot expect to build BrcmPatchRAM with the current SDK, as it requires an older SDK (I suspect you changed the SDK settings in the project), although BrcmPatchRAM2 will build with current SDK (official builds are with the settings as is in the project).

RehabMan, 2018-09-18:
As an example of my process, just today the App Store offered an update to Xcode 10.
Given their history, I suspected this would delete the 10.13 SDK and give me the 10.14 SDK instead.
So, before I did the update, I saved the 10.13 SDK to LegacySDKs:
```
sudo cp -R /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk /Applications/LegacySDKs/MacOS210.13.sdk
```
Then I allowed the update to Xcode 10, ran it one time, quit.
Then ran my legacysdks.sh script.
Which results in now having SDKs 10.16 -> 10.14 available in Xcode 10.

peterlobster, 2018-09-25:
@RehabMan Excellent explanation. It was exactly what I was messing for a while. I remember reading something at not too long ago on the Clover thread on InsanelyMac. If I'm not mistaken, you, Slice, and few other people where going back and forth about it. Just hasn't thought about it really but this is exactly what I needed. Thanks Rehabman!