RehabMan/OS-X-BrcmPatchRAM #57: BCM20702 under Mojave

unixb0y, 2018-10-16:
Hello, I wanted to explain how I got a USB dongle with BCM20702 working under Mojave.
I tried putting the kext files in /L/E, didn't work so after reading a different issue (from today actually), I tried /S/L/E. While that didn't work at first, I then tried to refresh the kext cache; the files didn't show up in `kextstat`.
I used this command: `sudo kextcache -invalidate /` which told me 2 error messages: `BrcmFirmwareStore` and `BrcmPatchRAM2` had the wrong permissions. This is fixed like so:

> sudo chown -R 0:0 /System/Library/Extensions/BrcmPatchRAM2.kext/
> sudo chmod -R 755 /System/Library/Extensions/BrcmPatchRAM2.kext/
> sudo chown -R 0:0 /System/Library/Extensions/BrcmFirmwareRepo.kext/
> sudo chmod -R 755 /System/Library/Extensions/BrcmFirmwareRepo.kext/

Reboot and be happy! 😄
Both the actual Bluetooth entry in System Preferences, that was missing before as well as an entry in System Report > Bluetooth finally appeared!
I have a TP-Link Archer T9E WiFi card and now in conjunction, AirDrop does work like a charm 👍

RehabMan, 2018-10-16:
All hack/non-Apple kexts go to /L/E, not /S/L/E.
But you must install them correctly.
There are procedures in my laptop FAQ:
http://www.tonymacx86.com/el-capitan-laptop-support/164990-faq-read-first-laptop-frequent-questions.html

closed 2018-10-16 by RehabMan

unixb0y, 2018-10-16:
It's not a laptop... :D