RehabMan/OS-X-BrcmPatchRAM #59: is it possible to use this patch to upload firmware for intel cnvi bluetooth?

steve3d, 2018-12-15:
I just got a gigabyte Z390 Aorus Master mainboard, which has a cnvi wifi+bluetooth, and I found that if I boot windows first, then boot into mac, the bluetooth works, has firmware, recognized.

but if I just code boot into mac directly, then the bluetooth is not recognized.

This CNVI bluetooth still connected with usb, so I wondered if there is any possiblilty of using this to upload the firmware to CNVi bluetooth?

steve3d, 2018-12-15:
by works I mean I can use bluetooth to pair with other bluetooth device, without any change or kext.

But I have to boot into windows first.

najdanovicivan, 2019-02-11:
I am also interested in that. I have intel Wireless-AC 3168 and how I currently use it is by booting into windows in virtual machine to load the bluetooth firmware

I've been looking at some linux sources 

https://elixir.bootlin.com/linux/v5.0-rc6/source/drivers/bluetooth/btusb.c#L1742
https://elixir.bootlin.com/linux/v5.0-rc6/source/drivers/bluetooth/btusb.c#L1628

Methods  btusb_setup_intel and btusb_setup_intel_patching

Method for writing the firmware on intel bluetooth uses different HCI commands compared to Broadcom

https://elixir.bootlin.com/linux/v5.0-rc6/source/drivers/bluetooth/btbcm.c#L114

So it won't work without modifying the code support intel cards

Anyway I was able to extract firmwares from windows driver I don't know if they are in correct format but there are always ones from linux available

Here are the windows driver firmwares
[FW Intel BT.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/2853133/FW.Intel.BT.zip)

RehabMan, 2019-04-17:
This kext is for Broadcom only.  Closing.

closed 2019-04-17 by RehabMan