RehabMan/OS-X-BrcmPatchRAM #71: macOS Catalina

headkaze, 2019-06-15:
Please check out my post on insanelymac [here](https://www.insanelymac.com/forum/topic/339175-brcmpatchram2-for-1015-catalina-broadcom-bluetooth-firmware-upload/)

DanaHan, 2019-06-15:
I have tried this as post said for DW1830. Good news is after modify the BrcmBluetoothInjector-Info.plist. The sidecar can find my iPad. But I can not connect any bluetooth device. The bluetooth fireware can not be loaded. The version is c4096. Any suggest? Thank you.

DanaHan, 2019-06-15:
Addition, I have solved this problem. Some post said, injector need to active bluetooth in windows. I tried to connect a mouse in windows and then in macOS everything is ok.