RehabMan/OS-X-BrcmPatchRAM #72: Updated firmware for DW1820A 0A5C:6412

RV-ABZ, 2019-06-30:
BrcmFirmwareRepo + BrcmPatchRAM2 offer an outdated firmware version that does not fully support Bluetooth services with Dell's DW1820A carrying PCIe id 0A5C:6412.

I've used a newer firmware version extracted from the latest driver I found on Dell's web site and, after updating the above 2 x kexts, Bluetooth services worked 100% Ok.

I've posted the details here:
https://osxlatitude.com/forums/topic/11540-dw1820a-for-7490-help/?do=findComment&comment=91833

However, I was not able to update the kexts so that SysInfo displayed firmware version v5803 instead of v5799. It's only cosmetic but I was wondering if you could update the kexts accordingly. I did not manage to successfully recompile the build under 10.14.

Is the actual FW version automatically passed to SysInfo or is it just hardcoded in the compiled kexts?

Many thanks,
Hervé

mishurov, 2019-07-17:
The latest can be extracted from here
https://www.dell.com/support/home/us/en/04/Drivers/DriversDetails?driverId=7X3JX

BCM4350C5_003.006.007.0145.2724 I believe

It won't solve the problem though. The version of the firmware which is currently in the repo works fine in Linux.

chatelp, 2020-02-05:
HI @RV-ABZ, I have the exact same question as you. Did you manage to make it work in the end? Using release 2.5.1 of RrcmPathRAM i get v7 c4689 firmware. You?

RV-ABZ, 2020-02-05:
Yes. See here:
https://osxlatitude.com/forums/topic/12392-solved-7490-catalina-dw1820a-bluetooth-problem/

NB: I did sort out the above firmware update matter in the end and was able to inject the latest version in the kexts.

chatelp, 2020-02-05:
It works ! Merci Hervé :) Do we know why this particular version works? 🤔