RehabMan/OS-X-BrcmPatchRAM #73: Firmware for DW1820A 0a5c.6412

DRenan, 2019-07-12:
Hello RehabMan! 

I've bee trying to make the DW1820A work on a new build with your set of BRCMxxx.kext and unfortunately the outcome has been disastrous. I've read on a post from Hervè that the firmware included for the device id is a very old one, and have ventured to try and add the latest I could find to BrcmFirmwareRepo.kext and then modifying the info.plist inside BrcmPatchRAM2.kext to match the new firmware but still the bluetooth device works real bad.

Production of the DW1560 is gone now and they are real hard to find, so my only choice is DW1820a, many users are starting to buy this module and are in the same situation as me.

I don't know (probably) if I am doing something wrong, thats why I am writing to you over here. If you need any data or anything please do let me know, I really want this module to work, and could be great for the community. thank you very much in advanced.

DRenan, 2019-07-17:
Hello again, 

I have successfully "compressed" the newer firmware to .zhx (with your set of tools) I could find in Dell website and managed to properly load the desired firmware, but Bluetooth still is not working..

it seems that some users are having success with BrcmBluetoothInjector.kext, but looking in your repo I could only find firmware injector for 0a5c_6412 FW v5779, I don't know how could I fill the firmware section inside the .plist with the correct data from the firmware I have, could you please point me to the right direction?

All I see is people complaining about the firmware not loading properly and not much information about it.

This is the outcome (be ware that I removed all other firmwares just so I can make sure nothing else is loading)
[Archive.zip](https://github.com/RehabMan/OS-X-BrcmPatchRAM/files/3402846/Archive.zip)

I wouldn't mind to pay for your service in order to compile a working set of kext for the card to work.

Thanks

chatelp, 2020-02-05:
HI @DRenan, I have the exact same question as you. Did you manage to make it work in the end? Using release 2.5.1 of RrcmPathRAM i get v7 c4689 firmware. You?

DRenan, 2020-02-05:
Hello @chatelp I ended up using the new set of kext built by Headkaze, you can see his post in Insanelymac, jus search for DW1820A, there is a stable version that works quite fine.

Good luck

chatelp, 2020-02-05:
Thanks @DRenan . In the end I found out about this other version : https://osxlatitude.com/forums/topic/12392-solved-7490-catalina-dw1820a-bluetooth-problem/
Do you think it's the same?