RehabMan/OS-X-BrcmPatchRAM #76: Unable to update firmware and no log shows.

youzhonghui, 2019-11-19:
This has been confused me for a few days. I am running 10.14 majave and using a bluetooth dongle 0a5c:21e8. The bluetooth icon shows but not working properly.  

I installed the BrcmPatchRAM2.kext and BrcmFirmwareRepo.kext to L/E with the hackintool. The bluetooth firmware stay v14 c4096, and there are no message from BrcmPatchRAM shows in /var/log/system.log. I also tried to copy BrcmFirmwareData.kext and BrcmPatchRAM2.kext to the EFI/Clover/kexts/Other, but still same result. 

I am not sure if I am doing something wrong. Can anyone help me? I will be veeeeeeeery grateful.  

[image]

[image]

freedomandcourage, 2019-11-19:
Same problem of Ram3.kext on catalina 15.1 ~

youzhonghui, 2019-11-20:
I partially solved my problem.

1. Delete all bluetooth kexts used to installed on clover/kexts/other and L/E
2. Download latest release from https://github.com/acidanthera/BrcmPatchRAM/releases
3. Install `BrcmPatchRAM3.kext`，`BrcmBluetoothInjector.kext` 和 `BrcmFirmwareRepo.kext` to the L/E with hackintool
4. Unplug the bluetooth dongle
5. Shutdown
6. Restart and plugin the dongle

and the firmware upgrade to c5860.

closed 2019-11-20 by youzhonghui